OPEnSLab-OSU/ednaServer #77: Modify offshoot clean for EPA

NathanJesudason, 2023-05-06:
**Describe the feature you'd like**
EPA needs an offshoot clean state that fully clears everything, Riley and Josh are discussing the exact method for this.

NathanJesudason, 2023-05-12:
First step is confirming the code version that EPA uses, the second step is then modifying the task procedure for that version based on Riley's and Kai's recommendation

princeri1, 2023-05-16:
The EPA has given the following Requirements:
1. Fully clear the inlet line of water between runs

It is estimated that the following changes to the code will have the ideal effect:

Branch: MasterV3
1. Change the time for the AirFlush State's time variable from 15 to 60 seconds